nicokaiser/rpi-audio-receiver #149: Bluetooth pairing instructions

ettiennegous, 2023-07-19:
Hi @nicokaiser , 
While setting this up myself I scoured the Wiki and Readme files but couldnt find good instructions on how to pair via bluetooth and also to ensure that if the Pi is rebooted that the previously connected device connects again automatically.

I can raise a PR in the readme to add more detailed BT instructions but figured I would ask first before I do as I might have missed it.
Is this documented in the repo/wiki at the moment?

As an example the commands I had to follow after installation scripts were

```
sudo bluetoothctl
power on
agent on
# Pair the phone, note down the MAC address 
trust 00:00:00:00:00:00 # Put Phone MAC address here so after reboot it can connect again
```

nicokaiser, 2023-07-20:
Thanks for your comment! I currently do not use Bluetooth at all, so I cannot tell if it works or not. I thought (!) the `bt-agent` process took care of automatically accepting any device – but I can't tell if this still works, and when read-only mode is active, this might lead to strange issues after reboot.
So, I'd really appreciate working instructions for Bluetooth devices.

mousemat86, 2023-07-23:
@ettiennegous jumping in here to say I had a similar experience when setting up bluetooth recently and I used similar instructions to what you've listed above, which worked for me.

Could I ask if you're experiencing volume issues when using bluetooth similar to this: https://github.com/nicokaiser/rpi-audio-receiver/issues/121

However in my case I'm using MacOS and I'm unable to adjust volume using the system, only in various players.

ettiennegous, 2023-07-24:
@mousemat86 I used it in the car paired with my phone I cant confirm that use case no.

ettiennegous, 2023-07-24:
@nicokaiser I am unable to raise a PR, can you please open up collaboration on the repo?

[image]

ettiennegous, 2023-07-25:
PR has been [raised here](https://github.com/nicokaiser/rpi-audio-receiver/pull/150)

nicokaiser, 2023-08-04:
Thanks! I merged #150

closed 2023-08-04 by nicokaiser