nicokaiser/rpi-audio-receiver #154: Both user and system pulseaudio systemd services are running

mbalasz, 2023-09-20:
Hi,

On a fresh Raspberry Pi 4 with (64-bit OS Lite) I ran your script and rebooted the device. I then checked systemd status for pulseaudio. However, it looks like both the system and the user services are up and running.

```
$ systemctl status  pulseaudio
● pulseaudio.service - Sound Service
     Loaded: loaded (/etc/systemd/system/pulseaudio.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-09-20 22:21:56 BST; 2min 47s ago
   Main PID: 414 (pulseaudio)
      Tasks: 2 (limit: 3933)
        CPU: 540ms
     CGroup: /system.slice/pulseaudio.service
             └─414 /usr/bin/pulseaudio --daemonize=no --system --disallow-exit --disable-shm --exit-idle-time=-1 --log-target=journal --realtime --no-cpu-limit
```

```
$ systemctl status --user pulseaudio
● pulseaudio.service - Sound Service
     Loaded: loaded (/usr/lib/systemd/user/pulseaudio.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2023-09-20 22:22:09 BST; 3min 25s ago
   Main PID: 775 (pulseaudio)
      Tasks: 2 (limit: 3933)
        CPU: 281ms
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/pulseaudio.service
             └─775 /usr/bin/pulseaudio --daemonize=no --log-target=journal
```

Is this expected? AFAIU, your script is supposed to mask the user-level PulseAudio service.

nicokaiser, 2023-09-21:
Yes `install.sh` should disable the user-level PulseAudio service, but apparently it does not.

Does running `systemctl --global mask pulseaudio.service` help, or does this break something else?

nicokaiser, 2024-01-04:
Planning to replace Pulse Audio with BlueALSA in #164

nicokaiser, 2024-01-05:
Fixed with #164

closed 2024-01-05 by nicokaiser