nicokaiser/rpi-audio-receiver #158: Install Failed:  autoreconf: automake failed with exit status: 1

claytonjacobs, 2023-10-13:
Ran the install instructions on my RPi4 (Linux rpi 6.1.21-v7).  Install script ended with this:

```
update-alternatives: using /usr/bin/automake-1.16 to provide /usr/bin/automake (automake) in auto mode
Setting up libglib2.0-dev:armhf (2.66.8-1) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for sgml-base (1.30) ...
Processing triggers for install-info (6.7.0.dfsg.2-6) ...
Processing triggers for libglib2.0-0:armhf (2.66.8-1) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u7) ...
Setting up libavahi-client-dev:armhf (0.8-5+deb11u2) ...
Setting up libpulse-dev:armhf (14.2-2+rpt1) ...
Cloning into 'alac'...
remote: Enumerating objects: 43, done.
remote: Counting objects: 100% (43/43), done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 43 (delta 7), reused 38 (delta 6), pack-reused 0
Receiving objects: 100% (43/43), 78.32 KiB | 673.00 KiB/s, done.
Resolving deltas: 100% (7/7), done.
aclocal: warning: couldn't open directory 'm4': No such file or directory
libtoolize: putting auxiliary files in '..'.
libtoolize: copying file '../ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:40: installing './compile'
configure.ac:63: installing './config.guess'
configure.ac:63: installing './config.sub'
configure.ac:20: installing './install-sh'
configure.ac:63: error: required file './ltmain.sh' not found
configure.ac:20: installing './missing'
Makefile.am: installing './INSTALL'
codec/Makefile.am: installing './depcomp'
autoreconf: automake failed with exit status: 1
```

claytonjacobs, 2023-10-13:
I can play audio through Bluetooth with no issues.

sborn, 2023-10-22:
I had the same thing happen just now on a completely clean install of Bookworm fully updated. I had run install-bluetooth.sh and it seemed fine. I got the error on install-shairport.sh 

I got the exact same errors, including the m4 directory error.

sborn, 2023-10-22:
if I edit out the alac section, it appears to work

nicokaiser, 2023-10-22:
Could you try to rename `install.sh` to something else and run it? I had a similar problem some time ago...
(Unfortunately I cannot test this on Raspbian 12 at the moment, this script was made for Rasbpian 11 Bullseye and may not run at all on Bookworm...)

sborn, 2023-10-23:
I just did that and it worked. Thanks! It is documented here: https://github.com/mikebrady/alac/issues/4

nicokaiser, 2024-01-05:
Fixed with #163

closed 2024-01-05 by nicokaiser