nicokaiser/rpi-audio-receiver #159: Cannot connect using Bluetooth USB dongle

MichaelDietzel, 2023-11-11:
Hi,

I am using a Raspberry Pi 4B that I already have for some while but I did a fresh install using the new Raspberry Pi OS based on debian 12 (64 bit version). First I attempted using the internal bluetooth but there was lots of stuttering (as expected). So I got an external CSR Bluetooth 5.0 USB dongle. The dongle is shown to be powered on in bluetoothctl. First I attempted just powering the internal bluetooth off (using bluetoothctl) and reconnecting my smartphone (deleting the old connection). The device took a lot more time than another (idle) bluetooth device next to it to show up on my phone and then attempting to connect took a long time as well. My phone asked me if I wanted to share my contacts and I said no but after that I only was connected (at least I saw the bluetooth connected icon (or maybe it just was a connecting icon?)) for a few seconds before I got disconnected. When playing some music before being disconnected I do not hear anything. Next I disabled the internal bluetooth in the config.txt and after a reboot there only was the external bluetooth left in bluetoothctl, but the results stayed the same. Also re-running the installscripts did not help.

So what could be the reason for this?
- Am I doing something wrong in the way how I try to use the external dongle? I didn't find any information in the Wiki on how to use it, I just found I should use one.
- Is the new Raspberry Pi OS not supported, yet? However the internal bluetooth at least works a little bit
- Could this be a bluetooth dongle incompatibility? At bluetoothctl sees it (do there is at least some driver support) and I also can see the device with my phone. Also I can some activity when pairing and watching what bluetoothctl reports during that.

Is there any more information I could provide that would help figuring out what is going on?

Thanks!

nicokaiser, 2023-11-20:
I am sorry, I have no idea. I am still using Raspberry Pi OS 11 and did not update any of the scripts for the new version 12. Since Bluetooth on Linux is still somewhat of a mess, I might even revert to BlueALSA on Debian 12, since the bluealsa package is included (and updated) on the current Debian.

nicokaiser, 2024-01-05:
Could you retry with the current script?

MichaelDietzel, 2024-10-15:
It is working now, thank you! Although I am using a different bluetooth stick now as well. I don't have access to the previous one any longer.