nicokaiser/rpi-audio-receiver #161: Error message while trying to connect: Pin oder Passkey wrong

georgar70, 2023-12-31:
Hello,
I'm trying to connect my Android Phone (Lineage 20.0). with the Raspberry (Bluetooth box installed) and I get the error message that the pin or passkey are wrong.
The following facts:
1. I have two Raspberry Bluetooth boxes
2. I have two phones (Galaxy A72, Lineage 20.0)
3. with my newly installed phone I get the error message when I try to connect to one of the two Raspberry boxes. When I connect to another Bluetooth device I do not get an error message
4. i have deleted the cache on the phone and reset the connections (WLAN, Bluetootth) to factory settings. I still get the error message.

Is there anything I can check/do on the Raspberry side?

Greetings

nicokaiser, 2023-12-31:
Are you sure you are using Raspberry Pi OS 11 (not 12) and have the `sspmode` set to `1` in the config? Using a PIN for connection does not really work: https://github.com/nicokaiser/rpi-audio-receiver/wiki/Bluetooth-PIN 

And, which USB Bluetooth dongle do you use? I tested some cheap dongles which did not support Bluetooth 4 and it did not work at all with them.

georgar70, 2023-12-31:
Thank you for your quick reply.
My answers to your questions:
1.I use this bluetooth dongle with both Raspberry bluetooth boxes:
USB Bluetooth 4.0 Adapter - Windows XP,Vista,7,8,10,Linux- Market and Technology
It has worked without any problems so far.
2. yes, I followed your instructions and installed PI OS 11.
The boxes have also been working well for a year with the second Galaxy A72 phone (same Lineage OS) and my son's phone (Xiaomi with Lineage 18) 
Only the devices that I now have don't work: my Galaxy Phone and a Google Chromecast. The Google Chromecast had a connection for a short time and when I changed the user and switched back, no connection was established.
My questions:
1. where do I set "sspmode set to 1"? If it was described in your instructions, I did it, but I can also check it. But the question still arises why it works with other devices.
2. how can I avoid a PIN being requested? I have not requested or set this anywhere.

nicokaiser, 2023-12-31:
> where do I set "sspmode set to 1"

This is being set in `install-bluetooth.sh`, so as long as you did not change anything there, it should work.

> how can I avoid a PIN being requested? I have not requested or set this anywhere.

Honestly, I have no idea. From what I have experienced, this might happen when the Bluetooth adapter does not support simple paring mode, but since you are using the same adapter on both devices, your adapter should be fine. With simple pairing mode, the client devices should not need any PIN to connect...

georgar70, 2023-12-31:
How can I ensure that the devices use the simple paring mode?
Could it be that you have to set something on Android? I haven't found any obvious paring settings even in the developer menu.

What I have noticed is that when both boxes are active, it is possible that the name of one box is displayed first and then switches to the name of the second box. However, this is not always the case.
For this reason, I tried paring with only one box switched on, but this also failed.

Is it worth trying to stop and restart the Bluetooth service? This should also be the case if I restart the Raspberry, right?

georgar70, 2023-12-31:
Hello,
just a thought: could it be that the MAC address of the phone is stored in a Bluetooth config in the Raspberry?
Reason for the question: The Bluetooth connection worked on my phone with the same OS. I only had the problem after reinstalling it.
Note: I have just installed LineageOS 20 on my daughter's phone: she also has no problems with the connection.

nicokaiser, 2023-12-31:
> could it be that the MAC address of the phone is stored in a Bluetooth config in the Raspberry

Yes, I think so, but I'm not sure. Since Bluetooth audio is always a big mess, I can only use trial-and-error here...

closed 2023-12-31 by nicokaiser

georgar70, 2024-01-02:
@nicokaiser tAssumption: the mac address of the connected devices is saved on the raspberry (I don't know where it is saved).
This leads to a conflict when the OS of the connected device is reinstalled.
Solution: Reinstall Bluetooth software on the raspberry. I have also reinstalled the raspberry system. After that you can pair all devices again without any problems