nicokaiser/rpi-audio-receiver #167: move Uninstallation up in the README

lolie145, 2024-01-05:
i didn't read all the way through (my own fault). And now i installed it without knowing it couldn't be uninstalled. a waring before the install step would be appreciated.

nicokaiser, 2024-01-05:
I’ll consider this. Although this should be common sense to not just blindly execute scripts without fully understanding what they do…

closed 2024-01-06 by nicokaiser

Siekerenterprises, 2024-02-01:
Hi
I am attempting to install you project and keep getting the error "Unsupported OS: Raspberry Pi OS 12 (bookworm) is required.
I imaged and loaded both the lite and the full, with the same result.
Am I missing something.

nicokaiser, 2024-02-01:
Could you retry with the current version?

Siekerenterprises, 2024-02-02:
Thanks for the reply, I will give it a try

commit 894ab819 (2024-01-06): docs: add note about uninstallation (fixes #167)