nicokaiser/rpi-audio-receiver #168: Audio doesn't work. Maybe because i'm using the integrated bluetooth chip?

leotecno09, 2024-01-12:
I know that u guys recommend to use an external bluetooth dongle, but (because i'm stubborn) i tried to use the built-in chip and the audio doesn't work. Is the bluetooth chip the cause?

leotecno09, 2024-01-13:
UPDATE: It works with the built-in bluetooth chip. My problem was that i was using the wrong audio interface.

StuartIanNaylor, 2024-02-11:
Its not that it doesn't work its that it doesn't work well simulataneousy with the wifi.
So if using ethernet it could be OK

nicokaiser, 2024-02-11:
It does work. It only causes lots of problems, from reduced reception to audio stuttering and crashes, so I stopped to recommend using the internal Bluetooth chip and cannot examine any related problems.

closed 2024-02-11 by nicokaiser