nicokaiser/rpi-audio-receiver #172: No Audio on ShairPort-Sync via 3.5mm

mabearce1, 2024-04-29:
So I have TRIED digging and diving into the internet and I cannot find the answer here. 

I followed the instructions to the T on the Rpi-Audio-Receiver and it has installed just fine and dandy. 
my phones, computers, etc find the AirPlay on the Raspberry Pi just fine and connects say its playing and I get no audio from it. I have tried digging and digging to try and fix this issue. But when I installed it I also installed the Spotify connect, and this works PERFECTLY via 3.5mm jack. 

Here is my config file when typing in --I think it's something to do with my ALSA settings? 
The ones on the internet have a MUCH larger config file but mine doesn't look this way? maybe the installer isn't installing it correctly? 

```
shairport-sync -X

>> Display Config Start.

From "uname -a":
 Linux Air 6.6.28+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.28-1+rpt1 (2024-04-22) aarch64 GNU/Linux

From /etc/os-release:
 Debian GNU/Linux 12 (bookworm)

From /sys/firmware/devicetree/base/model:
 Raspberry Pi 3 Model B Rev 1.2

Shairport Sync Version String:
 4.3.2-AirPlay2-smi10-alac-OpenSSL-Avahi-ALSA-soxr-sysconfdir:/etc

Command Line:
 shairport-sync -X

Configuration File:
 /etc/shairport-sync.conf

Configuration File Settings:
 general : 
 {
   name = "Air Pi";
   output_backend = "alsa";
 };
 sessioncontrol : 
 {
   session_timeout = 20;
 };

>> Display Config End.
>> Goodbye!
 
```

nicokaiser, 2024-04-30:
> The ones on the internet have a MUCH larger config file but mine doesn't look this way?#

The [default config](https://github.com/mikebrady/shairport-sync/blob/master/scripts/shairport-sync.conf) is not very much larger when removing all commented out lines.

You could try to figure out if shairport-sync tries to play on a different ALSA device than raspotify. `aplay -l` gives you a list of ALSA devices, and there should only be the 3,5mm one (otherwise you might want to disable unused devices like HDMI etc. to avoid confusion).

mabearce1, 2024-04-30:
So I ran that --- 
Looks like it's the Headphones and the HDMI. 

So I went into the config file and did the  "dtoverlay=vc4-kms-v3d,**noaudio**" adding in there. 

We shall see ill test when I get home 

`matt@Air:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0`

mabearce1, 2024-04-30:
Hey! We are working thanks for the tip! Disabling the HDMI audio worked for me. So ONLY headphones is an option! AirPlay and Spotify work flawlessly! 

Now....just need to see if we can get bluetooth Audio in to work...so I can connect my Google Home to it for whole home Google sync!