nicokaiser/rpi-audio-receiver #184: The Raspberry Pi Bluetooth hardware

seamusdemora, 2025-05-08:
Hello,

This looks like an interesting project; I may try it once I finish with the project I've begun :) 

I'm writing to ask a question re the statements you made re the RPi BT hardware. I think I may be up against something similar! 

Can you tell me whether or not the RPi BT hardware can support the A2DP profile? I **assumed** it could, but [I read earlier today](https://www.usa.philips.com/c-f/XC000011139/what-are-the-differences-between-bluetooth-versions) that while A2DP is supported by **_versions_** of BT since ver 1.2, that does **not** imply support for A2DP. Rather, **A2DP must be built in by the manufacturer**. 

I've learned from hard experience that the RPi organization is not forthcoming re limitations of their hardware. I've found nothing indicating their BT hardware supports A2DP, and so I now have begun to wonder. 

Anyway - I'm just curious to know what you might have learned....  and thanks for your time!

nicokaiser, 2025-05-09:
The Raspberry Pi (all versions) do support A2DP with their Bluetooth chipset. It is possible to transmit audio via Bluetooth to these devices.

**However**, the implementation is quite weak, since the same chip is used for Wi-fi. Especially when Wi-fi is enabled, you will run into all kinds of connectivity problems, dropped connections, etc. when trying to stream audio over Bluetooth to the on-board chip. All of these connectivity-related problems went away when using a proper external USB Bluetooth dongle.

So I stopped wasting my time trying to get decent A2DP over the internal Bluetooth chip. It's not worth it.

closed 2025-05-09 by nicokaiser

seamusdemora, 2025-05-11:
OK, thank you for the information - that helps.  And I share your views WRT BT/WiFi... I feel the majority of the issues are due to the use of a common antenna. Really brain-dead design.